Spark-Concepts/xPro-V5 #169: 4 axis setup. Changing A axis microsteps affects Z axis.

pjackson5, 2022-08-10:
Hi,
Ive just updated to the latest firmware. Im trying to set up a 4 axis machine.

These are the settings im using when i type $+ -

```
$0=5 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=1 (Invert limit pins, boolean)
$6=1 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=1 (Soft limits enable, boolean)
$21=1 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=1 (Homing direction invert, mask)
$24=100.000 (Homing locate feed rate, mm/min)
$25=1000.000 (Homing search seek rate, mm/min)
$26=250.000 (Homing switch debounce delay, milliseconds)
$27=10.000 (Homing switch pull-off distance, millimeters)
$30=24000.000 (Maximum spindle speed, RPM)
$31=0.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$N1=
$N0=
$33=5000.000
$34=0.000
$35=0.000
$36=100.000
$100=800.000 (X-axis travel resolution, step/mm)
$101=800.000 (Y-axis travel resolution, step/mm)
$102=800.000 (Z-axis travel resolution, step/mm)
$103=5.926
$104=100.000
$105=100.000
$110=3000.000 (X-axis maximum rate, mm/min)
$111=3000.000 (Y-axis maximum rate, mm/min)
$112=3000.000 (Z-axis maximum rate, mm/min)
$113=50.000
$114=1000.000
$115=1000.000
$120=600.000 (X-axis acceleration, mm/sec^2)
$121=600.000 (Y-axis acceleration, mm/sec^2)
$122=600.000 (Z-axis acceleration, mm/sec^2)
$123=600.000
$124=200.000
$125=200.000
$130=120.000 (X-axis maximum travel, millimeters)
$131=181.000 (Y-axis maximum travel, millimeters)
$132=88.000 (Z-axis maximum travel, millimeters)
$133=100000.000
$134=300.000
$135=300.000
$140=1.950
$141=1.950
$142=1.950
$143=1.950
$144=0.800
$145=0.800
$150=1.250
$151=1.250
$152=1.250
$153=1.250
$154=0.400
$155=0.400
$160=32
$161=32
$162=32
$163=32
$164=16
$165=16
$170=16
$171=16
$172=16
$173=16
$174=16
$175=16
```

So I have these loaded. Then i type $163=16 to change the microsteps on the A axis from 32 to 16. Not changing anything on the Z axis, it was already set up correctly.

I reset/turn it on and off again, check that $163 is set to 16

And now my **Z** axis moves twice as far. It moved 10mm before, now it moves 20mm. If all im doing is changing **A** axis microsteps it shouldnt do this right?

Spark-Concepts, 2022-08-13:
Would you mind pasting the link OR providing the name of the 4 axis firmware .bin file you are currently using.

pjackson5, 2022-08-13:
Hi Spark-Concepts, yes I should have specified im using -
CNC_xPRO_V5_XYAZ_NO.bin
Downloaded on the 9th

Spark-Concepts, 2022-08-17:
Pjackson5,

We're currently looking into this, I'll keep you posted

pjackson5, 2022-08-26:
Hi @Spark-Concepts thanks for looking into this. Were you able to reproduce the problem?

Jennywhite2, 2022-08-26:
@pjackson5 sir,you have Facebook or other social media, can you share your 4axis cnc router..🤗🤗

MattChristianNZ, 2022-11-28:
Any updates? I'm having the same problem.