Spark-Concepts/xPro-V5 #172: Settings do not persist

g1kw, 2022-08-15:
I have what appears to be an obscure issue.

Every time I power on my xPro or hit a hard limit or press reset in cncjs, I need to do a little dance of erasing and reapplying my settings for the cnc to work again. 

If I do not perform the following steps every time the motors wont engage for any axis.

Step 1 - Send $NVX and $RST
Step 2 - Press reset button on xPro-V5
Step 3 - Apply my configuration through cncjs
Now it will work until power off, hard limit or soft reset in cncjs.

I am hoping I am missing something obvious. Any ideas?

Spark-Concepts, 2022-08-15:
What version of CNCJS are you using? If you are using a more recent version CNCjs, I would suggest rolling back to the most stable version [1.9.22](https://github.com/cncjs/cncjs/releases/tag/v1.9.22)

spydercnc, 2022-08-15:
Use CNC 3D Commander as your sender. It stores the machine settings and a lot easier to set the xPro..

Craig

g1kw, 2022-08-17:
I reinstalled cncjs to 1.9.22 and tried again. Same obscure result whereby settings need to erase non-volitile memory and then reset.

I also noted this time that a job ran and I had left the machine, it must have gone in to standby/sleep state, the only way to get it "awake" again was to erase non-volatile memory and reset as above in my original post.

I will try out CNC Commander and verify the result is same or different.

Spark-Concepts, 2022-08-17:
Once you've applied all your settings, please provide the following:
1. Send ```$S``` - copy & paste the entire response
2. Send ```$V``` - also copy & paste the response

g1kw, 2022-08-17:
$S / $V shown below.

I did try CNC Commander as well and same result.

> $S
$Sta/SSID=CNC_xPRO_V5
$Sta/Password=******
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$AP/SSID=CNC_xPRO_V5
$AP/Password=******
$AP/IP=192.168.0.1
$AP/Channel=1
$System/Hostname=CNC_xPRO_V5
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=AP
$Bluetooth/Name=xPRO_V5_BT
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$Message/Level=Info
$User/Macro0=
$User/Macro1=
$User/Macro2=
$User/Macro3=
$Homing/Cycle0=Z
$Homing/Cycle1=XY
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Enable/Delay=0
$Stepper/Direction/Delay=0
$Stepper/Pulse=4
$Stepper/IdleTime=255
$Stepper/StepInvert=
$Stepper/DirInvert=Y
$Stepper/EnableInvert=On
$Limits/Invert=On
$Probe/Invert=On
$Report/Status=1
$GCode/JunctionDeviation=0.010
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=Off
$Limits/Hard=On
$Homing/Enable=On
$Homing/DirInvert=XY
$Homing/Squared=
$Homing/Feed=100.000
$Homing/Seek=1000.000
$Homing/Debounce=250.000
$Homing/Pulloff=10.000
$GCode/MaxS=24000.000
$GCode/MinS=8000.000
$Laser/FullPower=1000
$GCode/LaserMode=Off
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Coolant/Delay/TurnOn=1.000
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=PWM
$X/StepsPerMm=200.622
$Y/StepsPerMm=32.272
$Z/StepsPerMm=200.000
$A/StepsPerMm=32.272
$B/StepsPerMm=100.000
$C/StepsPerMm=100.000
$X/MaxRate=2500.000
$Y/MaxRate=2500.000
$Z/MaxRate=2500.000
$A/MaxRate=1000.000
$B/MaxRate=1000.000
$C/MaxRate=1000.000
$X/Acceleration=50.000
$Y/Acceleration=50.000
$Z/Acceleration=50.000<Idle|MPos:-995.065,-35.449,13.800|FS:0,0>
$A/Acceleration=200.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=300.000
$Y/MaxTravel=300.000
$Z/MaxTravel=300.000
$A/MaxTravel=300.000
$B/MaxTravel=300.000
$C/MaxTravel=300.000
$X/Current/Run=1.800
$Y/Current/Run=1.800
$Z/Current/Run=1.800
$A/Current/Run=1.800
$B/Current/Run=0.800
$C/Current/Run=0.800
$X/Current/Hold=0.250
$Y/Current/Hold=0.250
$Z/Current/Hold=0.250
$A/Current/Hold=0.250
$B/Current/Hold=0.400
$C/Current/Hold=0.400
$X/Microsteps=8
$Y/Microsteps=8
$Z/Microsteps=8
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
$Errors/Verbose=Off
ok

> $V
[MSG: NVS Used: 225 Free: 405 Total: 630]
ok

Spark-Concepts, 2022-08-17:
You have ```$Limits/Hard=On``` this setting uses the limit switches to "Alarm" and stop the machine in the event of an axis overrun.. 

If your using the limit switches for homing ```$Limits/Hard=Off```

g1kw, 2022-08-17:
Thanks for that tip. I do not use Homing generally, but I will additionally try this setting.

Would that explain that when in an Alarm state that I need to erase non-volatile, hit the reset switch on the xPro, and then reapply my settings?

Something I also noted whilst testing today is that the motors do not appear to be in their "hold" state (ie there is no slight buzz sound from the motors)  when the power to the xPro is turned on until reset button on the xPro is hit.

Spark-Concepts, 2022-08-17:
With $Limits/Hard=On, the switches are used to stop Grbl if it hits a limit switch. This is a good fail safe setup, but it does an immediate uncontrolled stop and you must rehome before you can use the machine again.

Spark-Concepts, 2022-08-17:
Also, if you haven't already, I would suggest updating with the latest firmware [CNC_xPRO_V5_XYYZ_NO.bin](https://github.com/Spark-Concepts/xPro-V5/blob/main/Firmware/CNC_xPRO_V5_XYYZ_NO.bin) 

Also update the webUI... see https://github.com/Spark-Concepts/xPro-V5/issues/166#issuecomment-1207468602

**note: click the green folder icon to update the index.html file**

closed 2022-08-25 by Spark-Concepts

g1kw, 2022-08-25:
None of the above actually fixed the issue. I just gave up and do my reset dance every time I use the machine. It sucks.