Spark-Concepts/xPro-V5 #173: xPro V5 and Huanyang VFD with 0-10V speed control

tomtonia, 2022-08-18:
Hi All,

I'm trying to setup Huanyang spindle control with 0-10V line coming out of the xPro V5 controller.

I had VFD setup for manual control and it worked fine and then I have first followed these instructions
[(https://github.com/Spark-Concepts/xPro-V5/wiki/Front_Panel#toolheadrs485)]
changed
`$Spindle/Type=PWM`
`$GCode/MaxS=24000`
and now between GND and 0-10V on the Toolhead port I get current roughly relating to speed selected so for M3 S24000 it's close to 10v for M3 S12000 ~5V
on the Huanyang VFD I have moved the jumper to the VI position and connected cables coming out of the Toolhead port to the VI(+) and ACM(GND) pins.
Like I said spindle was working fine with manual controls and the only VFD settings I have changed are below
PD001 - 0 (some say to change it to 1 but this requires additional input on FOR pin or bridging FOR and DCM to start the spindle)
PD002 - 1
PD070 - 0 (0-10V)
PD072 - 400 (Higher analog frequency) - my understanding is that this is the speed/frequency relating to max (10V) input
PD073 - 0 (Lower analog frequency) - as above but for 0V input

With all those changes I can control the spindle speed but it never stops. When I send M5 multimeter still reads 0.47V on the xPro Toolhead output which translates to ~1050RPM on the VFD
Basically 0-10V output never drops to 0V
I have already tried using fine tune pot but it doesn't make any difference to M5(OFF) state voltage and it always stays above 0.4V

It there something wrong with my setup or am I missing something?

Any help will be appreciated.

Thanks

ImagineerNL, 2022-08-19:
not sure if you are the same person as on the RatRig Facebook group yesterday where i posted this answer.

I have been struggling with this same issue with my Kress/AMB Digital spindle and managed to fix it for me today.
What you have to do is actually the same as is mentioned in the wiki ([Point 4 regarding PWM](https://github.com/Spark-Concepts/xPro-V5/wiki/Front_Panel#toolheadrs485).)

1. $Spindletype=PWM
2. Make sure the relay jumper inside the box is set to Spindle Enable (not Mist)
3. Wire the 0-10V pinout through the Relay to your 0-10V spindle wire.
4. I have the Kress 10-26V wired directly to the PSU, but with your VFD that might be different)
5. now, whenever you send an M3 Sxxxx command, the relay (Spindle enable) is opened. This ONLY applies when you send a speed as well (grbl safety)
6. When sending M3 S0 the signal wil remain on and the spindle wil keep having the 0.5V which is my lowest. So S0 is actually not zero. (Not sure if it helps when you set spindleMin to 3500 in my case)
7. Sending an M5 command closes the relay(spindle enable). I have this as one of my Macro's.

I will write a more extensive howto on a later time.
This is the only way i can get my AMB Digital spindle working through the xPro; for a VFD the RS485 option might be the best way to go; keeping in mind that for that route the relay is suggested to be used as well as a safety feature.

tomtonia, 2022-08-19:
Thanks for your reply. 
No, I'm a different person :-)

Your solution is something I was going to try today after seeing it somewhere else this morning while looking for a solution. 
It kinda makes sense to use that relay for enabling/disabling the VFD. I would rather keep it available for something else but this workaround might have to do for now.

This way of connecting VFD is suggested in the documentation but because this is suggestion you would think that your spindle should work without it.

> For SAFETY and unintentional actuation of the spindle when a PWM signal isn't being commanded, it is suggested that you wire the PWM through the relay and move the relay jumper(1) to SP_EN as illustrated below

ImagineerNL, 2022-08-19:
youre welcome.

There might be cases where the minimum voltage is low enough to not power on, but mine takes a minimum of 0.5V to power on, and i get 0.515 :(

Take note of the issue i just opened ( #174 ) when going this route.

Emsand1, 2023-02-04:
I know you said you wanted to avoid RS485 and I'm relatively new to this, but I haven't experienced any spindle related issues using RS485 and my Huanyang 1.5kw water-cooled spindle. It's performed well.