Spark-Concepts/xPro-V5 #178: XPro-V5 laser still running on S0

Mmmeng20, 2022-09-14:
Hi, recently added a laser to my Queenbee Pro, controlled by the XPro-V5. I'm slowly getting things figured out but am still having problems with the laser not completely turning off with an S0 speed input. The laser is wired through the relay so an M0 or M5 command is the only way to completely turn it off. This is fine for cutting but ruins any engravings. After doing some research on here it seems that S0 will still produce a 0.5V signal, which is what I'm seeing. The laser is a Sculpfun S9 and works perfectly with the gantry/controller it came with. Any suggestions? Thanks!

ImagineerNL, 2022-09-15:
What is the max signal it can have?
The 0-10V is tuneable with a screw inside the controller.
You can tune the output to be about 0.2V, but then you wont be able to reach the 10V, which is useless for a spindle with a full 0.5-10V range, but might be okay for your laser

Mmmeng20, 2022-09-15:
I am using the 0-5V PWM pin. With no load (laser not hooked up) I get an output of (0.5-0.7)-4.71V and (0.35-0.4)-4.71V with the laser hooked up. The output at S0 is pretty unstable, hence the range listed at the low end.

I wasn't sure if adjustment screw changed the 0-5V PWM output at all but I tried it anyway yesterday and found that it didn't change the voltage at all.

I also tested out the controller that came with the laser and it outputs a 1.4mV-4.97V signal