Spark-Concepts/xPro-V5 #179: Please update System/Control to also 'Shut Down'

ImagineerNL, 2022-09-15:
Currently $/System/Control=RESTART restarts the controller.

System/Control | ESP444 | Admin | RESTART | Restart the Grbl controller.  Requires value = RESTART

I seem to have to do this a lot of times immediately after a power-on.
This might be because of shutting down through a powercut is not great.
If we could power off with $System/Control=SHUTDOWN , we can add it in a macro for the system to do a clean powerdown, after which it is safe to cut the power.

synman, 2022-11-29:
I can weigh in here.  Technically, the only way to power down an ESP32 board is with a power switch.  However, it does have a couple sleep related modes that might be of interest.

https://randomnerdtutorials.com/esp32-deep-sleep-arduino-ide-wake-up-sources/

synman, 2022-11-29:
Here's an example of how we could implement this.  As noted in the video comments, I would release this with a deep sleep interval of like a year so the only way to get the board to respond would be by physically powering it off and disconnecting its USB connection / powering off the device connected to its USB port.  You could always just hit the reset button too.

https://youtu.be/vWcv7r3Q1N4

Here are the actual changes:  https://github.com/synman/Grbl_Esp32/commit/48ebdcc71c3d8226427fefea5a74a391147b716b