Spark-Concepts/xPro-V5 #183: Stepper motors not responding using xPro-V5

JRVDWALT, 2022-10-02:
Hi everyone,

I recently (2 months ago) bought the QueenBee Pro 1000x1000 from Bulkman which included the xPro-V5 controller.
I have managed to set everything up and everything was working like it should. I went on holiday and I reconfigured the speed of the stepper motors (lower) and try to jog the machine, but nothing happens. I have checked all the stepper motors and they are working. I have uninstalled everything and reinstalled but still nothing.

It seems like the motors is not getting power from the controller. The controller switches on and can connect via WIFI or USB.
No red light on the controller so I am a bit stuck. I use CNCjs and here is the configuration below:

$$
$0=4 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=1 (Step direction invert, mask)
$4=1 (Invert step enable pin, boolean)
$5=1 (Invert limit pins, boolean)
$6=1 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.020 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=0 (Homing direction invert, mask)
$24=50.000 (Homing locate feed rate, mm/min)
$25=635.000 (Homing search seek rate, mm/min)
$26=250.000 (Homing switch debounce delay, milliseconds)
$27=2.500 (Homing switch pull-off distance, millimeters)
$30=24000.000 (Maximum spindle speed, RPM)
$31=0.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$100=314.961 (X-axis travel resolution, step/mm)
$101=314.961 (Y-axis travel resolution, step/mm)
$102=314.961 (Z-axis travel resolution, step/mm)
$103=200.000
$104=100.000
$105=100.000
$110=2000.000 (X-axis maximum rate, mm/min)
$111=2000.000 (Y-axis maximum rate, mm/min)
$112=2000.000 (Z-axis maximum rate, mm/min)
$113=1000.000
$114=1000.000
$115=1000.000
$120=150.000 (X-axis acceleration, mm/sec^2)
$121=150.000 (Y-axis acceleration, mm/sec^2)
$122=150.000 (Z-axis acceleration, mm/sec^2)
$123=200.000
$124=200.000
$125=200.000
$130=782.000 (X-axis maximum travel, millimeters)
$131=752.000 (Y-axis maximum travel, millimeters)
$132=300.000 (Z-axis maximum travel, millimeters)
$133=300.000
$134=300.000
$135=300.000
ok

and here is some more

$S
$Sta/SSID=CNC_xPRO_V5
$Sta/Password=******
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$AP/SSID=CNC_xPRO_V5
$AP/Password=******
$AP/IP=192.168.0.1
$AP/Channel=1
$System/Hostname=CNC_xPRO_V5
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=AP
$Bluetooth/Name=xPRO_V5_BT
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$Message/Level=Info
$User/Macro0=
$User/Macro1=
$User/Macro2=
$User/Macro3=
$Homing/Cycle0=Z
$Homing/Cycle1=XY
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Enable/Delay=0
$Stepper/Direction/Delay=0
$Stepper/Pulse=4
$Stepper/IdleTime=255
$Stepper/StepInvert=
$Stepper/DirInvert=X
$Stepper/EnableInvert=On
$Limits/Invert=On
$Probe/Invert=On
$Report/Status=1
$GCode/JunctionDeviation=0.020
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=Off
$Limits/Hard=Off
$Homing/Enable=On
$Homing/DirInvert=
$Homing/Squared=
$Homing/Feed=50.000
$Homing/Seek=635.000
$Homing/Debounce=250.000
$Homing/Pulloff=2.500
$GCode/MaxS=24000.000
$GCode/MinS=0.000
$Laser/FullPower=1000
$GCode/LaserMode=Off
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Coolant/Delay/TurnOn=1.000
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=NONE
$X/StepsPerMm=314.961
$Y/StepsPerMm=314.961
$Z/StepsPerMm=314.961
$A/StepsPerMm=200.000
$B/StepsPerMm=100.000
$C/StepsPerMm=100.000
$X/MaxRate=2000.000
$Y/MaxRate=2000.000
$Z/MaxRate=2000.000
$A/MaxRate=1000.000
$B/MaxRate=1000.000
$C/MaxRate=1000.000
$X/Acceleration=150.000
$Y/Acceleration=150.000
$Z/Acceleration=150.000
$A/Acceleration=200.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=782.000
$Y/MaxTravel=752.000
$Z/MaxTravel=300.000
$A/MaxTravel=300.000
$B/MaxTravel=300.000
$C/MaxTravel=300.000
$X/Current/Run=1.800
$Y/Current/Run=1.800
$Z/Current/Run=1.800
$A/Current/Run=1.800
$B/Current/Run=0.800
$C/Current/Run=0.800
$X/Current/Hold=0.250
$Y/Current/Hold=0.250
$Z/Current/Hold=0.250
$A/Current/Hold=0.250
$B/Current/Hold=0.400
$C/Current/Hold=0.400
$X/Microsteps=16
$Y/Microsteps=16
$Z/Microsteps=16
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
$Errors/Verbose=Off
ok

I am fairly new to CNC so if it was something simple, I will apologise in advance.
I would appreciate any help thank you

Spark-Concepts, 2022-10-04:
I know this soulds silly, but please confirm you have 24V applied to the xPro and the power switch is (ON); the green power LED should also be illuminated. 

If that checks good, connect via USB using CNCjs then press the "Reset" button, and copy & paste the response.

JRVDWALT, 2022-10-04:
Hi Guys,

Thank you for the response. And yes 24v is applied to the xPro :-) with a green power LED

I have done what you asked and pressed the reset button and here is the response:

ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.3a Date 20220621]
[MSG:Compiled with ESP32 SDK:v3.2.3]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_NC]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Door switch on pin GPIO(16)]
[MSG:Macro Pin 1 GPIO(13)]
[MSG:Macro Pin 2 GPIO(0)]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000701]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(-782.000,0.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(-752.000,0.000)]
[MSG:Y  Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(-752.000,0.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-300.000,0.000)]
[MSG:Z  Axis Trinamic driver test passed]
[MSG:Initializing RS485 VFD spindle]
[MSG:Using Auto-RTS]
[MSG:VFD RS485  Tx:GPIO(4) Rx:GPIO(25) RTS:None]
[MSG:Local access point CNC_xPRO_V5 started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]
client> $$
[MSG:'$H'|'$X' to unlock]
$0=10 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=1 (Step direction invert, mask)
$4=1 (Invert step enable pin, boolean)
$5=1 (Invert limit pins, boolean)
$6=1 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.020 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=0 (Homing direction invert, mask)
$24=50.000 (Homing locate feed rate, mm/min)
$25=635.000 (Homing search seek rate, mm/min)
$26=250.000 (Homing switch debounce delay, milliseconds)
$27=2.500 (Homing switch pull-off distance, millimeters)
$30=24000.000 (Maximum spindle speed, RPM)
$31=8000.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$100=314.961 (X-axis travel resolution, step/mm)
$101=314.961 (Y-axis travel resolution, step/mm)
$102=314.961 (Z-axis travel resolution, step/mm)
$103=200.000
$104=100.000
$105=100.000
$110=2000.000 (X-axis maximum rate, mm/min)
$111=2000.000 (Y-axis maximum rate, mm/min)
$112=2000.000 (Z-axis maximum rate, mm/min)
$113=1000.000
$114=1000.000
$115=1000.000
$120=150.000 (X-axis acceleration, mm/sec^2)
$121=150.000 (Y-axis acceleration, mm/sec^2)
$122=150.000 (Z-axis acceleration, mm/sec^2)
$123=200.000
$124=200.000
$125=200.000
$130=782.000 (X-axis maximum travel, millimeters)
$131=752.000 (Y-axis maximum travel, millimeters)
$132=300.000 (Z-axis maximum travel, millimeters)
$133=300.000
$134=300.000
$135=300.000
ok
[MSG:Spindle RS485 Unresponsive 8]

I was troubleshooting and tried to connect the VFD Huang Yang last night to see if CNCjs will allow it to Spin but with no result.

Thank you again for your time and looking forward to hear back from the team.

Spark-Concepts, 2022-11-09:
We're you able to resolve your issue?... and yes, since the RS485 (spindle) is unresponsive the machine will be in a constant alarm state... RS485 operation 

1. Have you updated to the latest firmware? 
2. In addition you'll need to confirm your VFD is properly configured: https://github.com/Spark-Concepts/xPro-V5/issues/60

I've closed this issue, however please feel free to reopen if you're still having troubles

closed 2022-11-09 by Spark-Concepts