Spark-Concepts/xPro-V5 #184: Xpro V5 stops at 25% and says job complete

Todderbod, 2022-10-04:
Xpro v5 When I run my Gcode that I used aspire to make it get right around 25% and says job complete. The spindle is still running but is says job complete. This is only the second project I have tried the first one completed with no problem. This is a bigger file I'm not sure if that is the issue any help would be greatly appreciated

Spark-Concepts, 2022-10-04:
Are you saving the job file to the xPro's SD card via the WebUI? If so, please download the cut file to your PC and compare it to the original file (ensure it didn't get corrupted during the transfer). Please let me know what you find.

Todderbod, 2022-10-06:
The file is exactly the same . I would really like to figure this out it sucks spending this kind of money to have a paper weight

Spark-Concepts, 2022-10-07:
Todderbod,

Please post your settings and build information here (```$S``` and ```$I```). Also, what are you using to generate your .NC gcode cut files as this is most often the culprit?

Also, cnc3d commander is not a sender maintained by Spark-Concepts. I would ask that you attempt to run the job via the native xProV5 web-ui(wi-fi) and/or via CNCjs(USB) and monitor for similar issues (unplug the stepper outputs connectors while running the gcode file)?

For additional troubleshooting please use the contact-us link found here at www.spark-concepts.com 
...from here we can evaluate your gcode, and monitor the job on our test rig and provide you feedback (findings and solutions)

spydercnc, 2022-10-09:
Wow, I find the comment about CNC3D quite astounding to say the least. It may not be supported by SC but then again neither are many many of its clients/buyers. At least CNC3D do try (usually successfully)  to support people with the ongoing problems with the xPro 5, especially in regards to getting it to talk to a VFD. 
I am still waiting for a response from yourselves about a problem with my unit, where updating the firmware on the the Xpro, basically bricked my machine. The Webui, is at best pretty useless and it takes multiple attempt to connect to it -again still waiting for help with that from months ago.

Todderbod, 2022-10-10:
Spydercnc Does cnc3d have a firmware for the xpro5 that works better.  I think my whole problem is noise interference. Im going to redo all my setup and see what happens my controller is pretty close to my vfd. If I redo everything and I still have these problems I guess I'm going to try a new controller. Im rather new to cnc and getting pretty frustrated. I made a whole new gcode with aspire and ran it through cncjs now it stops spindle keeps going and computer still says its moving but its not. I watched a video that said a surge protector would work. Well guess what not for me same issue. Thats why Im going to redo my wiring job and get the vfd as far away from the xpro as I can.

spydercnc, 2022-10-11:
No it is just sender software, but was easier to reload an older version of firmware for the VFD, that worked.

Spark-Concepts, 2022-10-11:
@spyder - I was simply stating that we do not test and confirm compatibility with commander.   That's all,  we test and confirm CNCjs so that is what's suggest for troubleshooting to eliminate variables.   Have you submitted a support form on your unit (link to contact us page)

closed 2022-11-09 by Spark-Concepts