Spark-Concepts/xPro-V5 #185: Y Stepper Output Malfunctioning

Emsand1, 2022-10-09:
CNC xPro_v5 recently purchased. Set up and was running fine with several projects completed. Was homing the machine before my next job and the Y stepper motor locked up. It sounded like it was fighting itself.
Ton of troubleshooting including:
1.	Checking and swapping new wire leads.
2.	Checked belt tension.
3.	Checked extrusion for debris, damage.
4.	Disassembled motor/pulleys/wheels/brackets.
5.	Swapped motor with known working motor.
6.	Re-flashed XYYZ NO firmware.
7.	Swapped Y & A (y2) motor connections at box for testing the pair and each separately.

Any motor run from the Y connection port did the same locking. The replacement Y motor worked fine on A/y2 connection port but not Y connection port. I tested out with a multi-meter, but I think I read that isn’t accurate. Either way, It looked like A/y2 output was 24v on one phase and 4-5 volts on the other. Y connection had 24v & .9 volts.

$Sta/SSID=Sandel
$Sta/Password=******
$Sta/IPMode=Static
$Sta/IP=192.168.68.59
$Sta/Gateway=192.168.68.1
$Sta/Netmask=255.255.254.0
$AP/SSID=CNC_xPRO_V5
$AP/Password=******
$AP/IP=192.168.0.1
$AP/Channel=1
$System/Hostname=CNC_xPRO_V5
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=STA
$Bluetooth/Name=xPRO_V5_BT
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$Message/Level=Info
$User/Macro0=
$User/Macro1=
$User/Macro2=
$User/Macro3=
$Homing/Cycle0=Z
$Homing/Cycle1=XY
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Enable/Delay=0
$Stepper/Direction/Delay=0
$Stepper/Pulse=10
$Stepper/IdleTime=255
$Stepper/StepInvert=
$Stepper/DirInvert=
$Stepper/EnableInvert=Off
$Limits/Invert=Off
$Probe/Invert=On
$Report/Status=1
$GCode/JunctionDeviation=0.010
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=Off
$Limits/Hard=On
$Homing/Enable=On
$Homing/DirInvert=XY
$Homing/Squared=
$Homing/Feed=100.000
$Homing/Seek=1000.000
$Homing/Debounce=250.000
$Homing/Pulloff=4.000
$GCode/MaxS=24000.000
$GCode/MinS=0.000
$Laser/FullPower=1000
$GCode/LaserMode=Off
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Coolant/Delay/TurnOn=1.000
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=NONE
$X/StepsPerMm=80.000
$Y/StepsPerMm=80.000
$Z/StepsPerMm=320.000
$A/StepsPerMm=80.000
$B/StepsPerMm=80.000
$C/StepsPerMm=80.000
$X/MaxRate=2000.000
$Y/MaxRate=2000.000
$Z/MaxRate=2000.000
$A/MaxRate=2000.000
$B/MaxRate=2000.000
$C/MaxRate=2000.000
$X/Acceleration=300.000
$Y/Acceleration=300.000
$Z/Acceleration=300.000
$A/Acceleration=300.000
$B/Acceleration=300.000
$C/Acceleration=300.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=780.000
$Y/MaxTravel=775.000
$Z/MaxTravel=65.000
$A/MaxTravel=775.000
$B/MaxTravel=300.000
$C/MaxTravel=300.000
$X/Current/Run=2.500
$Y/Current/Run=2.500
$Z/Current/Run=2.800
$A/Current/Run=2.500
$B/Current/Run=0.800
$C/Current/Run=0.800
$X/Current/Hold=0.500
$Y/Current/Hold=0.500
$Z/Current/Hold=0.500
$A/Current/Hold=0.500
$B/Current/Hold=0.400
$C/Current/Hold=0.400
$X/Microsteps=16
$Y/Microsteps=16
$Z/Microsteps=16
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
$Errors/Verbose=Off

$$
$0=10
$1=255
$2=0
$3=0
$4=0
$5=0
$6=1
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=3
$24=100.000
$25=1000.000
$26=250.000
$27=4.000
$30=24000.000
$31=0.000
$32=0
$100=80.000
$101=80.000
$102=320.000
$103=80.000
$104=80.000
$105=80.000
$110=2000.000
$111=2000.000
$112=2000.000
$113=2000.000
$114=2000.000
$115=2000.000
$120=300.000
$121=300.000
$122=300.000
$123=300.000
$124=300.000
$125=300.000
$130=780.000
$131=775.000
$132=65.000
$133=775.000
$134=300.000
$135=300.000

markcagatandavis, 2022-10-22:
I have the exact same problem! I got a complete unit replacement because of this issue. I ran a job for 4 min on the replacement unit and it occurred again. Y-axis port only... 

I had a sparky check all the wiring and nothing should be shorting out. Reading through forums, this is apparently really common with xPro v5. What's the go? Can this be fixed, or is the port fried and need a replacement unit?

markcagatandavis, 2022-10-22:
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.3a Date 20220621]
[MSG:Compiled with ESP32 SDK:v3.2.3]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_NC]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Door switch on pin GPIO(16)]
[MSG:Macro Pin 1 GPIO(13)]
[MSG:Macro Pin 2 GPIO(0)]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000701]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(-780.000,0.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(0.000,780.000)]
[MSG:Y  Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(0.000,780.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-110.000,0.000)]
[MSG:Z  Axis Trinamic driver test passed]
[MSG:PWM spindle Output:GPIO(25), Enbl:GPIO(4), Dir:None, Freq:5000Hz, Res:13bits]
[MSG:Client Started]
[MSG:Connecting Enter The Muktrix_2.GHz]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connected with 192.168.20.155]
[MSG:Connected with 192.168.20.155]
[MSG:Start mDNS with hostname:http://CNC_xPRO_V5.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]
>

markcagatandavis, 2022-10-24:
Do they respond to this?

Spark-Concepts, 2022-11-09:
Please use the contact us link found here & we'll take care of this issue: https://www.spark-concepts.com/contact-us-1/

closed 2022-11-09 by Spark-Concepts