Spark-Concepts/xPro-V5 #186: Limit switch leds always on

Ynitial, 2022-10-28:
I can't get my Openbuilds limit switches to work. The leds are always on and nothing happens when I trigger them manually. I've tried the different `$Limits/Invert` settings but that doesn't change anything. If I'm correct, the limit switch logic jumper should be on 5V.

These are my settings. Does anyone have any advice?

```
$0=4 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=1 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=0 (Homing cycle enable, boolean)
$23=3 (Homing direction invert, mask)
$24=200.000 (Homing locate feed rate, mm/min)
$25=2000.000 (Homing search seek rate, mm/min)
$26=250.000 (Homing switch debounce delay, milliseconds)
$27=2.500 (Homing switch pull-off distance, millimeters)
$30=24000.000 (Maximum spindle speed, RPM)
$31=0.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$100=200.000 (X-axis travel resolution, step/mm)
$101=200.000 (Y-axis travel resolution, step/mm)
$102=200.000 (Z-axis travel resolution, step/mm)
$103=200.000
$104=100.000
$105=100.000
$110=2500.000 (X-axis maximum rate, mm/min)
$111=2500.000 (Y-axis maximum rate, mm/min)
$112=2500.000 (Z-axis maximum rate, mm/min)
$113=1000.000
$114=1000.000
$115=1000.000
$120=50.000 (X-axis acceleration, mm/sec^2)
$121=50.000 (Y-axis acceleration, mm/sec^2)
$122=50.000 (Z-axis acceleration, mm/sec^2)
$123=200.000
$124=200.000
$125=200.000
$130=300.000 (X-axis maximum travel, millimeters)
$131=300.000 (Y-axis maximum travel, millimeters)
$132=300.000 (Z-axis maximum travel, millimeters)
$133=300.000
$134=300.000
$135=300.000
```

ImagineerNL, 2022-10-28:
You havent enabled softlimits: $20=1
Also, on my setup (ratrig killerbee) the homing direction invert mask $21=1

```
$0=4  ;  Step pulse time, microseconds
$1=255  ;  Step idle delay, milliseconds
$2=0  ;  Step pulse invert, mask
$3=2  ;  Step direction invert, mask
$4=0  ;  Invert step enable pin, boolean
$5=1  ;  Invert limit pins, boolean/mask
$6=1  ;  Invert probe pin, boolean
$10=1  ;  Status report options, mask
$11=0.010  ;  Junction deviation, millimeters
$12=0.002  ;  Arc tolerance, millimeters
$13=0  ;  Report in inches, boolean
$20=1  ;  Soft limits enable, boolean
$21=0  ;  Hard limits enable, boolean
$22=1  ;  Homing cycle enable, boolean
$23=1  ;  Homing direction invert, mask
$24=100.000  ;  Homing locate feed rate, mm/min
$25=1000.000  ;  Homing search seek rate, mm/min
$26=250.000  ;  Homing switch debounce delay, milliseconds
$27=5.000  ;  Homing switch pull-off distance, millimeters
$30=25000.000  ;  Maximum spindle speed, RPM
$31=300.000  ;  Minimum spindle speed, RPM
$32=0  ;  Laser-mode enable, boolean
$100=200.000  ;  X-axis steps per millimeter
$101=200.000  ;  Y-axis steps per millimeter
$102=200.000  ;  Z-axis steps per millimeter
$103=200.000  ;  unknown
$104=100.000  ;  unknown
$105=100.000  ;  unknown
$110=2600.000  ;  X-axis maximum rate, mm/min
$111=2400.000  ;  Y-axis maximum rate, mm/min
$112=1500.000  ;  Z-axis maximum rate, mm/min
$113=2400.000  ;  unknown
$114=1000.000  ;  unknown
$115=1000.000  ;  unknown
$120=200.000  ;  X-axis acceleration, mm/sec^2
$121=200.000  ;  Y-axis acceleration, mm/sec^2
$122=200.000  ;  Z-axis acceleration, mm/sec^2
$123=200.000  ;  unknown
$124=200.000  ;  unknown
$125=200.000  ;  unknown
$130=530.000  ;  X-axis maximum travel, millimeters
$131=740.000  ;  Y-axis maximum travel, millimeters
$132=120.000  ;  Z-axis maximum travel, millimeters
$133=740.000  ;  unknown
$134=300.000  ;  unknown
$135=300.000  ;  unknown
```

Ynitial, 2022-11-03:
I managed to solve it through PB with @ImagineerNL, thanks! Was nothing software related. I mirrored all emergency switch cables by accident. When I connected them correctly, the issue was gone.

closed 2022-11-03 by Ynitial