Spark-Concepts/xPro-V5 #192: Undefined VFD_RS485_RTS_PIN

Rakishaa, 2022-11-10:
Hope you are well .

I have the issue connecting xpro V5 with vfd hy02d223b .
Have separated RS cable from rest of the cables to avoid interference . 
Spindle can be controlled manually but not via RS485 port.
Would be grateful for any kind of advice.
Many thanks .

Settings on VFD: 
pd1=2
pd2=2
pd5=400
pd163=1
pd164=1
pd165=3

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.3a Date 20201212]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_RS485_NO]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000701]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(0.000,740.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(0.000,1240.000)]
[MSG:Y  Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(0.000,1240.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-100.000,0.000)]
[MSG:Z  Axis Trinamic driver test passed]
[MSG:Initializing RS485 VFD spindle]
[MSG:Undefined VFD_RS485_RTS_PIN]
[MSG:VFD RS485  Tx:GPIO(4) Rx:GPIO(25) RTS:GPIO(0)]
[MSG:Spindle RS485 Unresponsive 8]
[MSG:Critical Spindle RS485 Unresponsive]
[MSG:Local access point CNC_xPRO_V5 started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Flood coolant on pin GPIO(21)]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]
[MSG:Check limits]
[MSG:'$H'|'$X' to unlock]
> $$
$0=4 (Step pulse time, microseconds)
$1=25 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=1 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=1 (Hard limits enable, boolean)
$22=0 (Homing cycle enable, boolean)
$23=3 (Homing direction invert, mask)
$24=200.000 (Homing locate feed rate, mm/min)
$25=500.000 (Homing search seek rate, mm/min)
$26=250.000 (Homing switch debounce delay, milliseconds)
$27=2.500 (Homing switch pull-off distance, millimeters)
$30=24000.000 (Maximum spindle speed, RPM)
$31=24000.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$N1=
$N0=
$100=402.000 (X-axis travel resolution, step/mm)
$101=400.000 (Y-axis travel resolution, step/mm)
$102=200.580 (Z-axis travel resolution, step/mm)
$103=200.000
$104=100.000
$105=100.000
$110=4000.000 (X-axis maximum rate, mm/min)
$111=4000.000 (Y-axis maximum rate, mm/min)
$112=2500.000 (Z-axis maximum rate, mm/min)
$113=1000.000
$114=1000.000
$115=1000.000
$120=300.000 (X-axis acceleration, mm/sec^2)
$121=300.000 (Y-axis acceleration, mm/sec^2)
$122=100.000 (Z-axis acceleration, mm/sec^2)
$123=200.000
$124=200.000
$125=200.000
$130=740.000 (X-axis maximum travel, millimeters)
$131=1240.000 (Y-axis maximum travel, millimeters)
$132=100.000 (Z-axis maximum travel, millimeters)
$133=300.000
$134=300.000
$135=300.000
ok
>

FlyerLife22, 2022-11-11:
Share some pictures of your wiring?  The wiring on the HY manual is backwards for the RS+ and RS- , follow the ones here.  

What lights are blinking under the PWM - RS485 switch?  Red is xPRO talking,  Green is VFD talking  

The undefined RTS is ok, this controller uses an auto-RTS control.

Spark-Concepts, 2022-11-12:
Rakishaa,

The first things first... let's do a few things:
- Update your [firmware](https://github.com/Spark-Concepts/xPro-V5/tree/main/Firmware) to [CNC_xPRO_V5_XYYZ_NO.bin](https://github.com/Spark-Concepts/xPro-V5/blob/main/Firmware/CNC_xPRO_V5_XYYZ_NO.bin) 
- Once updated, set the spindle type _(spindle type will have been set to "none" after update)_ - in the command window send the following: ```$Spindle/Type=HUANYANG``` if that doesn't work, try ```$Spindle/Type=H2A```
- Ensure RS485 wiring is less than 1 meter in length, twisted, and connected properly (resistors are optional)
[image]
- Verify the **PWM-RS485** switch<sub>(1)</sub> is set to **RS485**
- Verify both the red and green LED's below the PWM-RS485 switch are flashing periodically
- Verify your VFD is configured properly using the procedure in the [Huanyang-VFD-manual](https://github.com/Spark-Concepts/xPro-V5/files/6247012/Huanyang-VFD-manual.pdf) _PLEASE REFER TO THE MANUAL SPECIFIC TO YOUR MODEL VFD; not all VFD's are the same despite the name on it_

Set the following register values via the VFD touchpad (should be the same on most VFD's - VERIFY ON YOUR SPECIFIC MODEL VFD):
| Register  | Value | Description |
| :-------: | :-----: | ------------- |
 | PD000 | 0 | **unlock parameters** |
 | PD001 | 2 | **Command source is RS485** |
 | PD002 | 2 | **Speed source is RS485** |
 | PD163 | 1 | **Communications address 1** |
 | PD164 | 1 | **9600 b/s** |
 | PD165 | 3 | **8 Bit No Parity - RTU** |
 | PD000 | 1 | **lock parameters** |
 
 _Also check the following settings (these will setting will vary based on the values indicated on the motor's nameplate and your mains voltage 110VAC vs 220VAC)_
 
| Register  | Description |
| :-------: | ----- |
| PD004 | **Base frequency as listed on spindle** _(typically 400)_ |
| PD005 | **Maximum frequency Hz** _(typical value for spindles is 400)_ |
| PD011 | **Min speed** _(recommended air-cooled=120 water=100)_ |
| PD014 | **Acceleration time** _(test to optimize)_ |
| PD015 | **Deceleration time** _(test to optimize)_ |
| PD023 | **Reverse run enabled** _(set to 1)_ |
| PD141 | **Spindle max rated voltage** _(typically 220)_ |
| PD142 | **Max rated motor current** _(0.8kw=3.7, 1.5kw=7.0, 2.2kw=??)_ |
| PD143 | **Motor poles** _(typically 2 or 4)_ |
| PD144 | **Rated motor revolution at 50Hz** _(typically 3000 @ 50Hz and 24000 @ 400Hz)_ |
 
- If you're still believe your having issues and trying to narrow down ground loop or issues otherwise (partial messages received etc..), load the following RS485 debug firmware: [CNC_xPRO_V5_XYYZ_NO_485_debug.bin](https://github.com/Spark-Concepts/xPro-V5/blob/main/Firmware/CNC_xPRO_V5_XYYZ_NO_485_debug.bin)
  - The RS485 Debug bin file is for debugging VFD RS-485 communication issues and requires USB connection and CNCJS or USB serial sender program
  - this firmware assumes/requires a NO (normally open) e-stop (unplug estop button after debug firmware is loaded)
  - Once connected to the X-Pro using CNCJS will echo commands sent to and from the VFD via RS-485
  - The terminal window will display messages from the X-Pro as Tx:....... and messages recieved from the VFD as Rx:........

In some rare cases where VFD noise is particularly severe (typically only encountered with VFD's on 220V mains); you may need to install an RS485 isolator - see https://github.com/Spark-Concepts/xPro-V5/issues/164#issuecomment-1219037733
   [image]

cdo1955, 2023-03-08:
I followed the above step with the RS 485 amplifier and still had intermittent issues controlling the spindle.
Until I put the Spindle controller in an metal electrical box and grounded the exterior of the box. The box has a slot opening
in the bottom I left open for air flow. Everything works now for the first time. The box is listed below.
[image]

Yolvan, 2023-03-30:
Herewith my setup with the twisted wire and the RS485 isolation module (see attached photos). Also attached are screenshots of the errors when I try and run the spindle with the M3 command. The spindle does not start from the current setup.
Please advise what to do next.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]