Spark-Concepts/xPro-V5 #194: Y axis faulty

danatomy, 2022-11-25:
Seems to be a recurring issue but posting here again as I'm not getting any repsonse from Sparks.

Just been testing my newly built Queenbee pro.  

The Y axis doesn't seem to function properly on my controller. Stepper makes a loud crunching noise and doesn't turn on properly therefore jamming the machine as the Y2 axis does work. 

I have ruled the stepper being faulty as I have plugged it into another port and it works perfectly. Any stepper that is plugged into this port does them same thing. 

I have checked all the wiring and thats fine. There is no fault light on. Updated the firmware too. 

I'm at a loss as what to do. Having trsted and read previous issues, I can't really see that it can be anything other than a fault with the controller. 

I would really appreciate a reponse from Spark concepts on this.

Sly75012, 2022-11-26:
I have the same issue, pls help to solve!

laufla, 2022-11-26:
Hi,
i had the same problem a few weeks after i received my box.

My contact at Spark was Mike  : ***@***.***

Deneteus, 2022-11-28:
> Seems to be a recurring issue but posting here again as I'm not getting any repsonse from Sparks.
>
> Just been testing my newly built Queenbee pro.
>
> The Y axis doesn't seem to function properly on my [...]

Typically if the firmware settings for the axis are right if there is a loud grinding noise on only one axis then that means the stepper driver was damaged. Unplugging a stepper in the midst of it running will do that and so will moving an axis manually while a stepper is still plugged in. Always test each stepper on a known good axis. Its hard to kill a stepper but very easy to kill a driver.

danatomy, 2022-11-28:
> > Seems to be a recurring issue but posting here again as I'm not getting any repsonse from Sparks.
> > Just been testing my newly built Queenbee pro.
> > The Y axis doesn't seem to function properly on my controller. Stepper makes a loud crunching noise and doesn't turn on properly therefore jamming the machine as the Y2 axis does work.
> > I have ruled the stepper being faulty as I have plugged it into another port and it works perfectly. Any stepper that is plugged into this port does them same thing.
> > I have checked all the wiring and thats fine. There is no fault light on. Updated the firmware too.
> > I'm at a loss as what to do. Having trsted and read previous issues, I can't really see that it can be anything other than a fault with the controller.
> > I would really appreciate a reponse from Spark concepts on this.
> 
> Typically if the firmware settings for the axis are right, and if there is a loud grinding noise on only one axis then that means the stepper driver was damaged. Unplugging a stepper in the midst of it running will do that and so will moving an axis manually while a stepper is still plugged in. Always test each stepper on a known good axis. Its hard to kill a stepper but very easy to kill a driver.

I didnt do either of these things.

ImagineerNL, 2022-12-03:
I have had this exact issue and had contact with Mike as well, through the above contact.

It MIGHT have something to do with issue #121 which is still unfixed ; What was your Current/Run set to?

Ynitial, 2022-12-03:
I'm currently finishing my build and found out my controller shows the same defect. I switched the stepper motors to double check that if the fault is in the controller and not the stepper motors. Used yellow and blue tape for clarity: https://photos.app.goo.gl/dnYzhLRggw9JCcms7

@ImagineerNL, my current/hold = 1.1 and current/run = 1.8, exactly as default by Spark wiki.

danatomy, 2022-12-04:
I'm now in contact with Mike. My current/run is the same, 1.8. Will see if we can resolve it.

synman, 2022-12-07:
Wouldn't ya know it, the exact same thing just happened to me, while in the middle of a job!

https://user-images.githubusercontent.com/1299716/206103446-869d8390-040a-41f7-93aa-ed9cae5a58fd.mp4

deviousdexter-original, 2022-12-08:
Hi - i have the same issue on a Ultimate Bee and am not getting anything from Mike after the initial response. Has anyone had any updates or resolutions to this matter?
many thanks

synman, 2022-12-08:
I emailed them (@Spark-Concepts ) at the same time I posted here -- and have not heard back from them (him).

I screwed up on my initial email which is why you see the completely unrelated issue tagged here.  

I've also reached out to Bulkman 3D as they were the reseller.  They asked I wait for Spark to respond and if I don't hear back to let them know.

ImagineerNL, 2022-12-08:
I have mailed Ratrig Sales in the beginning of October with this issue; They directed me to @Spark-Concepts to speed up the issue-finding.
After some back and forth I have received a new controller.
I have had some follow-up questions (e.g. 'what can I do to make sure this doesnt happen again?' and some other github-issue related questions), but it has been a radio silence since then, even after multiple requests.

Currently my new controller works flawless (and seemingly even better, but that can also be due to a lot of tweaking and tuning of the gantries while trying to pinpoint the issue). 
I'm just hoping my current controller doesnt bug on me again.

synman, 2022-12-08:
Tagging other occurrences of this (I stopped after page #2):

#185 
#152 
#147 
#146 
#144 
#140 
#139 
#133 
#127

MikeDMFC, 2022-12-19:
Same fault. Brand new. Y axis port started stalling, really great for twisting rails this is. Waiting response. I cant trust this xpro v5.

synman, 2022-12-22:
I have yet to hear a word from Spark and am in contact with Bulkman3d to see if I can get a replacement through them.

synman, 2022-12-22:
now I feel bad.   I had a box from Spark Concepts waiting for me in my mailbox tonight.

MikeDMFC, 2022-12-22:
> now I feel bad. I had a box from Spark Concepts waiting for me in my mailbox tonight.

Give it a good hard test run and let me know, i am sceptical of the reliability, on a dual axis it can be so damaging. I am waiting bulkman's managers response on mine

synman, 2022-12-22:
yeah.. that's part of my problem.  I am already running a different controller and am not in a hurry to reproduce the crash that happened on my Y axis when the xpro v5 gave up the ghost.

I may use it on a much smaller scale laser rig I run.

I also grabbed one of these:  https://github.com/zapta/simple_stepper_motor_analyzer

so I'm whole again, at least no worse than I was when all this started.  And I have a means to monitor current coming out of this thing now too.

Too bad it bit me and I went else where.  I was really excited about this controller when I first saw it.

Spark-Concepts, 2023-01-11:
This issue has been resolved - in the meantime, if you are having this issue please use the contact-us link found [here](https://www.spark-concepts.com/contact-us-1/)

Please note we are currently in the process of completely restructuring our support team and engineering staff to make many much needed improvements in order to drastically improve customer support and response.

closed 2023-01-11 by Spark-Concepts

synman, 2023-01-11:
What was done to actually resolve this issue?  I understand replacement units have been provided, but what was root cause and is this something we still have to watch out for?

I have zero trust in this controller right now.  Any info you can provide regarding the underlying issue would be greatly appreciated.  I do not see a new firmware or anything like that available so I'm kinda at a loss as to what was done to fix it.

Ynitial, 2023-01-13:
> What was done to actually resolve this issue?  I understand replacement units have been provided, but what was root cause and is this something we still have to watch out for?
>
> I have zero trust in [...]

For me, it was fixed by moving to a Blackbox X32.

RatRig (reseller) sent my email address to Spark Concepts weeks ago, which Spark asked because they apparently wanted to talk to me about this issue. Haven't heard from them at all.

Glad I moved to something else! This lack of communication and response to issues bothers me a lot. Even if they are in the dark with this: just let people know what's happening!