Spark-Concepts/xPro-V5 #196: pwm always at 5v

IdahoCreator, 2022-12-13:
Trying to run a laser with pwm. I get 4.9v ant both the 100% power(s100) and 50%(s50). It seems that when I turn on the laser I always get full power from laser. I dont have it connected to relay, will that help

synman, 2022-12-23:
have you installed a firmware / machine file that supports PWM.  Which firmware file are you using?

IdahoCreator, 2022-12-24:
I belive it has the newest version from the firmware file. The $32 is set af 100 and the voltage is 4.9v but to get the voltage to start lowering I habe to set the m3 or m4 below 10 and then we have a small wimdow for power adjustments. I have temporarly set $32 to 1000 and light burn maximum power to 100 to get any variation in pwm signal

synman, 2022-12-24:
There are easily 8 variations of the firmware though depending on your setup.  Which one did you flash?