Spark-Concepts/xPro-V5 #197: xPro V5 - Lost Connection

Yotapower, 2022-12-16:
Hi all,

I have a workbee cnc that I originally purchased with an NVUM controller, I was having issues with it and opted to switch to an xPro when it was released. I've had it about a 9 months, set it up and confirmed it was working but haven't used the machine very much until recently. I ran a profile cut to surface my spoilboard the other day and had no issues at all. Last night I ran a pocketing toolpath with no issue, but when I went to run the next toolpath, the spindle started but would not continue with the rest of the path. I wasn't able to pause or shut off the spindle, even with the E-stop. 

I was never able to get the wifi to connect to it and use the WebUI, so I just use the USB cable.

I'm not able to zero my work coordinates or jog the machine, it's like it's not even connected. I tried upgrading to the latest version of CNCJS, restarting everything and even a new usb cable. When it locked up initially, I was getting messages about RS485 unresponsive. 

Right after it happened when I loaded up CNCJS, I get the following in the command window:

CNCjs 1.9.27 [Grbl]
Connected to COM4 with a baud rate of 115200
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.3a Date 20210816]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_NC]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Door switch on pin GPIO(16)]
[MSG:Macro Pin 1 GPIO(13)]
[MSG:Macro Pin 2 GPIO(0)]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000703]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(-1290.000,0.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(-1290.000,0.000)]
[MSG:Y  Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(-1290.000,0.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-92.000,0.000)]
[MSG:Z  Axis Trinamic driver test passed]
[MSG:Initializing RS485 VFD spindle]
[MSG:Auto-RTS in use]
[MSG:VFD RS485  Tx:GPIO(4) Rx:GPIO(25) RTS:None]
[MSG:Client Started]
[MSG:Connecting TP-Link_A9A2]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:VFD: VFD settings read: RPM Range(7999, 24000)]]
[MSG:Connecting...]
[MSG:Connected with 192.168.0.174]
[MSG:Start mDNS with hostname:http://CNC_xPRO_V5.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]
client> $$
[MSG:'$H'|'$X' to unlock]
client> $X
client>
client> $X

When I tried to updated CNCJS, I just downloaded the newest version and ran the .exe file while I was connected to the xPro via USB. Am I not doing this correctly?

Now when I open CNCJS, it will not automatically connect. When I try selecting COM4 (Silicon Labs) from the port list, nothing happens. There is no blue light on on the xPro. 

I'm desperately trying to finish up some Christmas presents on my machine and this has really thrown a wrench in my plans. I have reached out to Mike @ Spark Concepts but figured I would also post here while waiting to hear back from him. 

Any assistance is greatly appreciated, thanks in advance!

synman, 2022-12-23:
I don't see anything wrong here.  Those messages are the standard startup messages the xPro spits out when you connect to it via USB and/or restart it.

As a matter of fact, the xPro is connecting just fine to your TP-Link_A9A2 access point / wireless router and is assigned the ip address 192.168.0.174.

You should be able to access it on our local network as http://192.168.0.174

If your computer supports mDNS (Windows w/ apple drivers installed, all Macs) it should be as simple as connecting to http://CNC_xPRO_V5.local

Beyond that, I'm seeing all of the stepper drivers initialize fine and your VFD's RS485 connection looks good too.

You are definitely going about connecting to the xPro V5 w/ CNCjs correctly.  It only supports USB connections.  

After connecting successfully with CNCjs, have you tried hitting the red reset button?  Often times CNCjs needs a grbl reset to start polling the connected machine.

Yotapower, 2022-12-23:
Unfortunately, I can no longer get those start up messages to come up. It's set to start up automatically, but now nothing happens when I open CNCjs and when I select silicon labs from the com port drop down nothing happens. 

I have never actually been able to use the wireless connection, I can't remember the specifics of it as I gave up trying. 

I have tried rebooting, resetting, unlocking and pressing the e-stop button many times in various combinations with no improvement. 

Thanks for the assistance, I got a response from Mike @ spark concepts today but no resolution. Hopefully he can figure something out.

SylvainBD, 2023-04-05:
Hello @Yotapower 
I have a similar problem with my xprov5 card. Not possible to connect it. Did you find an issue ?
Thanks for answer

Yotapower, 2023-04-05:
@SylvainBD No, I never figured it out. I purchased a OpenBuilds X32 to replace it and haven't had any issues since. 
I did find that +/- power terminals going into the box were all melted even though I was using 16ga wiring from the power supply, I'm guessing something shorted internally, I will be sending the box back to Spark Concepts for inspection and hopefully a refund. Generally pretty disappointing.