Spark-Concepts/xPro-V5 #199: Keep disconnected and stop in mid cutting.

nguyen4186, 2023-01-12:
Hi, I'm new to cnc. I just done with putting up my first workbee 1015 with xpro v5. Everything seem fine and moving correctly when jogging the axis, and turn on the spindle. My first problem is when using cncjs, the USB connection randomly lost after I sent m3 code, sometime it keep for few minutes but eventually dropped. Second, so I thought I could be driver problem and switched to using webui, and it all fine until I started running a cut, it will abruptly stop moving but with the spindle still on, and seem like it happened when the z axis plunging down. Please help,  I'm pretty new to all of this and any advice will greatly appreciated.

Andre1965ru, 2023-01-15:
замени cncjs на

OpenBuilds CONTROL

spydercnc, 2023-01-15:
Try using CNC3D commander for your sender, far more stable and designed to work with the xPro, plenty of support from their Admin as well.🙂

nguyen4186, 2023-01-19:
Thanks you for all your respond. I tried Commander of cnc3d and the same thing happened, it stop mid way and said job completed. Email Spark-Concepts and was asked to send in some pictures of my setup and gcode files, sent 4 days ago and still no response back. Waiting on new Blackbox controller and try it out now.

ZajtiM, 2023-01-21:
I know this is very basic, but might be worth to check. How do you connect your USB to xPro? Do you use USB C cable or you use micro USB to USB C adapter? I have USB C cable and adapter. And if i use adapter, i have problems with connection (bad adapter). But with USB C cable it works perfectly.

nguyen4186, 2023-01-21:
I do used a usb-c cable without an adapter, and if it a cable problem then it shouldn't happen when I run the job on SD card via webui. I got mine a blackbox x32 controller last night, plugged it in and now everything work just like it should, so my xpro v5 probably a defective one.