Spark-Concepts/xPro-V5 #201: VFD RS485 doesnt work and cannot upgrade firmware

tsamisa, 2023-02-03:
I  purchased an XPRO V5 via ratrig. The axis move properly but i couldn't control vfd via rs485. I followed the recommended guide and used two 120R resistors. I set the switch to the RS485 position and made the recommended changes to VFD and xpro. I tried with both $Spindle/Type=HUANYANG and $Spindle/Type=H2A
I get the following errors on cncjs
[MSG:Initializing RS485 VFD spindle]
[MSG:Undefined VFD_RS485_TXD_PIN]
[MSG:Undefined VFD_RS485_RXD_PIN]
[MSG:Undefined VFD_RS485_RTS_PIN]
[MSG:RS485 VFD spindle errors]
I tried to upgrade the firmware using the webui but i get an Error(3) File write error.
[image]

tsamisa, 2023-02-03:
After switching everything off and off i managed to upgrade to the new CNC_xPRO_V5_XYYZ_NO.bin. Now there is another problem that i keep getting an alarm that the limit switches are pressed. Keep in mind that before the upgrade there was no such issue and the limit switches were working fine.

vexedmillworkanddesign, 2023-02-24:
No positive, but as I am trying to get my machine up and running, I've been doing a lot of reading on issues. I believe your switches could be wired normally open, and the newer version requires them to be normally closed. Hope it helps!

tsamisa, 2023-02-24:
> No positive, but as I am trying to get my machine up and running, I've been doing a lot of reading on issues. I believe your switches could be wired normally open, and the newer version requires them to be normally closed. Hope it helps!

Thanks for the answer. What solved the false positives was indeed connecting the limit switches using both NC and NO pins as they suggest at their webpage. I don't know what changed after the upgrade but since it works now I'm ok with it