Spark-Concepts/xPro-V5 #202: How to set up 4th axis

cdo1955, 2023-02-07:
I am trying to find out the official recommended way to set up a 4th axis on my Queenbee CNC?
 Are there official instruction that you can provide for the xPro V5 controller?

akita11, 2023-02-19:
The default FW is for XYYZ, so we must flash a FW for XYZA. See https://github.com/Spark-Concepts/xPro-V5/wiki/Checking_firmware_and_upgrading .

VitaliiSadovshikov, 2023-03-10:
> so we must flash a FW for XYZA
Hi! Could I ask you? If my CNC utilizes two motors for Y-axis, i.e. all four motor ports on the controller are occupied, is it feasable to set up a supplementary rotary axis?

VitaliiSadovshikov, 2023-03-10:
> 

Hi! Have you implemented your setup with a 4th axis? And do you have two or one motors for Y-axis?