Spark-Concepts/xPro-V5 #207: Cannot Get Spindle To Turn On with RS485 and VFD

Volks1, 2023-04-11:
Hello, 

I am having an issue with getting the spindle to turn on with M3 S8000 command via the X-Pro Controller. I can get it to run fine in manual mode off the VFD, but I get this error message when I send the command:

CNCjs 1.9.22 [Grbl]
Connected to COM3 with a baud rate of 115200
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.3a Date 20220621]
[MSG:Compiled with ESP32 SDK:v3.2.3]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_NC]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Door switch on pin GPIO(16)]
[MSG:Macro Pin 1 GPIO(13)]
[MSG:Macro Pin 2 GPIO(0)]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000701]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(0.000,740.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(0.000,740.000)]
[MSG:Y  Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(0.000,740.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-80.000,0.000)]
[MSG:Z  Axis Trinamic driver test passed]
[MSG:Initializing RS485 VFD spindle]
[MSG:Using Auto-RTS]
[MSG:VFD RS485  Tx:GPIO(4) Rx:GPIO(25) RTS:None]
[MSG:Client Started]
[MSG:Connecting FiOS-TJQMJ]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connected with 192.168.1.232]
[MSG:Start mDNS with hostname:http://cnc_xpro_v5.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]
client> $$
[MSG:'$H'|'$X' to unlock]
$0=4 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=1 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=1 (Invert limit pins, boolean)
$6=1 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=3 (Homing direction invert, mask)
$24=2000.000 (Homing locate feed rate, mm/min)
$25=2000.000 (Homing search seek rate, mm/min)
$26=250.000 (Homing switch debounce delay, milliseconds)
$27=5.000 (Homing switch pull-off distance, millimeters)
$30=24000.000 (Maximum spindle speed, RPM)
$31=8000.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$100=400.000 (X-axis travel resolution, step/mm)
$101=400.000 (Y-axis travel resolution, step/mm)
$102=400.000 (Z-axis travel resolution, step/mm)
$103=400.000
$104=100.000
$105=100.000
$110=4000.000 (X-axis maximum rate, mm/min)
$111=4000.000 (Y-axis maximum rate, mm/min)
$112=4000.000 (Z-axis maximum rate, mm/min)
$113=4000.000
$114=1000.000
$115=1000.000
$120=300.000 (X-axis acceleration, mm/sec^2)
$121=300.000 (Y-axis acceleration, mm/sec^2)
$122=300.000 (Z-axis acceleration, mm/sec^2)
$123=300.000
$124=200.000
$125=200.000
$130=740.000 (X-axis maximum travel, millimeters)
$131=740.000 (Y-axis maximum travel, millimeters)
$132=80.000 (Z-axis maximum travel, millimeters)
$133=740.000
$134=740.000
$135=740.000
ok
client> $X
[MSG:Caution: Unlocked]
ok
[MSG:Spindle RS485 Unresponsive 10]
feeder> M3 S8000
[MSG:Critical Spindle RS485 did not reach speed 8000. Reported speed is -1 rpm.]
ok
ALARM:10
Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]
>

I have grounded the RS485 cable with the 100ohm resistors as stated in other fixes, and it is also grounded from the wire shielding. I tried a twisted pair of shielded Cat6 and that did not work either in the same setup. I have attached photos of the resistor and wiring setup. I have tried this with the debug and standrad firmware. Same issue either way also. I have also attached my VFD and CNC Settings in a Google Doc so I didn't make this message a mile long. 

Thank you for any assistance, I am pulling my hair our for a few weeks now :)

- Paul 

[CNC Settings.xlsx](https://github.com/Spark-Concepts/xPro-V5/files/11199049/CNC.Settings.xlsx)
[VFD Controller.xlsx](https://github.com/Spark-Concepts/xPro-V5/files/11199050/VFD.Controller.xlsx)
[image]
[image]
[image]

Volks1, 2023-04-11:
Hello,

It is a 2.2kw water cooled spindle. Min speed is 8000, or so it stated to set the spindle speed too.

Okay, I am open to trying any software to get some success. However why would one program make it work vs another if it wont turn on to begin with with the M3 command?

Volks1, 2023-04-11:
Okay I have the machine set up with CNC3D Commander. Very cool program, will definitely be using it going forward. 

Have uploaded the queen bee settings from thiprogram, and motion is fine but again the spindle does not turn on with any commands. Using the program on function and also sending an M3 code

spydercnc, 2023-04-11:
Have you changed the settings on the VFD to control it other than the controls on the VFD itself. There are a few lines of coding that need to be altered.

Volks1, 2023-04-11:
I have attached my VFD settings in an excel sheet in the original post

audiopavillon, 2023-04-17:
Hi Volks1

I think you must set the VI / VR VFD Jumper to VI !

Vi / VR Jumper is here 
[image]

audiopavillon, 2023-04-17:
>  Would this be a noise issue then?

It seems so, try to keep the RS485 cable as far as possible from the spindle cable.