Spark-Concepts/xPro-V5 #208: Upgrading firmware is failing

audiopavillon, 2023-04-14:
Hi everyone,

I attempt to upgrade the stock firmware to CNC_xPRO_V5_XYYZ_NC.bin 
I use the OTA process, at the end I get a "File write failed" Error(3) Message .

Any clue ?

TIA
Alain

spydercnc, 2023-04-14:
I have only ever successfully changed the firmware on xPro5 using CNC3D commander. The other times it also froze halfway through and bricked it. I had to reload a previous version to get it to work with the VFD anyway. Left it like that and has always worked well...

audiopavillon, 2023-04-15:
Hi spydercnc,

I am using Mac mini and macos Monterey ...
I get a problem similar as tsamisa, issue #201 

As I understand it I have to find a PC, and use CNC3D Commander ...

audiopavillon, 2023-04-15:
Hi Spydercnc,

Can you give me an howto to update the firmware with CNC3D Commander ?
Did you use CNC3D ESP32 Tool ?

I installed Parallels/Windows 11 on my Macmini but I can't connect to the machine properly, I will try to borrow a PC.
[image]

audiopavillon, 2023-04-16:
OK, I read the CNC3D docs.

spydercnc, 2023-04-16:
A very helpful site. Did you read the FW How-to? It is the same process as the Nighthawk.

spydercnc, 2023-04-16:
You can download our FW tool from here.
[Https://www.cnc3d.com.au/nhc](https://www.cnc3d.com.au/nhc?fbclid=IwAR1fjTXOCX6icbZ9W-jXdZ4kbI44QJr5oXi7IV_fQPyxpj_tEwsYhvv8o9g)

audiopavillon, 2023-04-16:
Hi Spydercnc,

Yes I have downloaded it, but as I previously wrote CNC3D commander seems to be unusable on my Macmini and Parallels Desktop, due to connection time out.
So I will borrow or buy a Windows laptop, and try again.

audiopavillon, 2023-04-17:
Hi everybody,

Solved !
The problem is a download one.
The process, you have to open a new page from the firmwares main page :
https://github.com/Spark-Concepts/xPro-V5/tree/main/Firmware

In my case it leads to :
https://github.com/Spark-Concepts/xPro-V5/blob/main/Firmware/CNC_xPRO_V5_XYYZ_NC.bin

There click the Download button, you will get a 1,9Mo firmware, that nicely get uploaded by XProv5 WebUI.

I have previously tried this way, firmwares main page, right-click on the CNC_xPRO_V5_XYYZ_NC link, download the file, the result was I got a 142ko file ... 

The first and second screen copies of the right way to process, the third is the wrong one. 
I'm a Mac user, don't know if the same thing could happen with Windows.

[image]
[image]
[image]