Spark-Concepts/xPro-V5 #209: unable to connect to the CNC xPro V5

JamesjoGibson, 2023-04-15:
New to the CNC world and trying to set up CNC xPro V5.
I can not connect the computer - CNCjs to the CNC xPro V5 via USB connection
When asked to select a com port I get one option 
Com 3
Manufacturer Silicon Labs
console opens with this message 
- CNCjs 1.10.2 [Grbl] connected to COM3 with baud rate 115200
 I can not enter any commands or access the jog controls to move anything
what am i doing wrong 
James

audiopavillon, 2023-04-15:
Checkbox "enable hardware flow control" -> x

JamesjoGibson, 2023-04-16:
We got it to connect Finn ally by restarting computer with cnc pro 5connected and on. 
But it’s very intermittent. 
connects only bout every 5 tries. 
The jog controls come up sometimes. ( not every time we connect)
Resetting on pro 5 returns message below and stays there 
[image]
Then can’t get jog controls to work
[image]

audiopavillon, 2023-04-16:
Hi,

I should try to unconnect all the cables but power and USB.
Must be connected every times.

You could try UGS, Openbuilds Control, etc ...