Spark-Concepts/xPro-V5 #213: RS485 Undefined - Spindle errors

johnhaytko, 2023-06-01:
This is typically a noise issue generated by the VFD:
- Ensure RS485 wiring is less than 1 meter in length, twisted, and connected properly (resistors are likely needed in your case)
[image]

In cases where VFD noise is particularly severe (typically only encountered with VFD's on 220V mains); you may need to install an RS485 isolator - see https://github.com/Spark-Concepts/xPro-V5/issues/164#issuecomment-1219037733
   [image]

_Originally posted by @Spark-Concepts in https://github.com/Spark-Concepts/xPro-V5/issues/193#issuecomment-1324432964_

johnhaytko, 2023-06-16:
> If you're running FluidNC - the RS485 pin descriptions are backwards in
> the example config.yaml file. If you're running the previous firmware then
> it will be something else.

I finally got the RS485 isolator installed. Issue is resolved. I
believe the integrators and resellers of these systems need to just ship
this $15.00 part with the product. The amount of time and aggravation with
this issue is ludicris.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Crypton10, 2024-10-16:
> I finally got the RS485 isolator installed. Issue is resolved. I believe the integrators and resellers of these systems need to just ship this $15.00 part with the product. The amount of time and [...]

Its a disgrace so many people having this issue Im still fighting with mine and cant get it to work . Sparks Concepts refuses to offer any support its been days since i contacted them with no answer. I have heard this from others it seems they no longer support this product . No one should buy this controller moving forward