Spark-Concepts/xPro-V5 #227: I face a problem in the x axis path

ghalib1399, 2023-07-28:
Hi All
When using the CNC machine, I face a problem in the x axis path, when the machine start to move, the machine gets stuck (freeze) for few seconds or less then continues the path, this freezing changes the x axis path and therefor the work is wrong. This problem always occurs in the x axis but sometime in the y axis as well. What is the solution ?

*QueenBee PRO
*CNC xPRO v5

https://github.com/Spark-Concepts/xPro-V5/assets/98682791/c0b2c191-357d-4b4e-912d-f23425da42fd

spydercnc, 2023-07-28:
Slow the feed down.
Make sure the machine is square and parallel. I actually had the two Y-axis about 10mm out of alignment and it solved my issue.
Remove the VFD wires from the drag chain, as it causes issues.
Make sure that the extractor is earthed. 
Put Ferrite chokes on everything to prevent RFI (interference)

ghalib1399, 2023-07-28:
> 

Can you explain what you mean by (put Ferrite chokes on everything to prevent RFI) everything= every wire?X,Y,Z ....

spydercnc, 2023-07-28:
All the power wires mainly, including the VFD, main incoming power cable, extractor, coolant pump etc.

ghalib1399, 2023-07-29:
Thank you for your help
I think I found the problem ! LeadScrews was loosen so I tighten it then the freezing gone

ghalib1399, 2023-07-31:
the problem not from LeadScrews :(