Spark-Concepts/xPro-V5 #234: What a huge mistake buying a XproV5

Crypton10, 2024-10-16:
Has numerous issues, and company no longer supports it. Anyone looking to buy this garbage please beware. first problem tickets sent in 4 days ago then sent another 2 days ago with no answer

spydercnc, 2024-10-16:
Join CNC Commander on FB. their software supports the Xpro.

Crypton10, 2024-10-16:
I have been getting some help from CNC commander they are great people and i am using their software but the xprov5 will not control the spindle and Im pretty sure its a hardware issue since i tried both rs485 and running with the 10v connection, Ill get the multi meter on it and see what the v5 is actually producing

Spark-Concepts, 2024-10-16:
Hi Crypton10,

Thank you for reaching out, and we apologize for the frustration you’ve experienced. Please note that all customer support issues have been moved to a dedicated platform, which you can access here: [spark-concepts.com/xpro-support](https://www.spark-concepts.com/xpro-support/) or at [ x-solutions.net](https://x-solutions.net/). 

Unfortunately, the issues posted on GitHub are no longer actively monitored and are kept for informational purposes only.
You can also view your current tickets by navigating to [MakerStore.help](https://www.makerstore.help/) and entering your email address. This will give you access to your past inquiries and responses from our dedicated support team.

Additionally, you can find the support link on our main website: https://www.spark-concepts.com/.

We encourage you to submit your support request through the provided link so we can assist you directly and resolve any issues you’re facing as quickly as possible.

Thank you for your understanding,
Bill
Spark Concepts Support Team

Crypton10, 2024-10-17:
Hi Bill thank you and Tawanda for the reply its very much appreciated ,I finally got the xprov5 to turn the spindle on in the software using the 10v instructions you provided only problem I seem to be having is its not recognising any of my speed input controls so when I start it up even though I have speed set at 8000 it ramps up to 24000 and it will not go down with any commands but I can turn it off, I checked the Voltage at the 10 v with the potentiometer  and Im only getting 1v dc on my multimeter Im using a fluke so I know its pretty accurate. Not sure if thats part of the problem ? All my cables are sheilded , sheild drained and grounded I also have ferrite on both ends of each cable to minimise any EMI just fopr reference
Anthony Magrini